FreeCAD/FreeCAD-Bundle #346: Linux fcitx input method

asperg, 2024-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The appimage for Linux lacks the library usr/plugins/platforminputcontexts/libfcitx5platforminputcontextplugin.so, which is why in systems using the fcitx input method you can only enter English characters.
Please add the library to appimage. Of course I can rebuild appimage myself but maybe it will help someone in the future

### Full version info

```shell
OS: Gentoo Linux (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * 3D_Printing_Tools
  * freecad.gears 1.2.0
  * fasteners 0.5.24
  * Curves 0.6.51
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-11-26:
@adrianinsaval could you take a look at this?

asperg, 2024-11-27:
I looked, I don't understand what actions are expected of me. Can you write more precisely?