FreeCAD/FreeCAD-Bundle #348: FreeCAD daily PPA builds are failing (2024-12-03)

furgo16, 2024-12-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I think this change (https://github.com/FreeCAD/FreeCAD/pull/17888) broke the daily PPA builds: https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-daily?field.series_filter=noble

An extract of the build log:

> Failed to copy 'src/Gui/Icons/freecad.xpm': No such file or directory at /usr/share/dh-exec/dh-exec-install-rename line 70, <> line 2.

Full build log: https://launchpadlibrarian.net/761762352/buildlog_ubuntu-noble-amd64.freecad-daily_0.22~pre1~202412032147~ubuntu24.04.1_BUILDING.txt.gz

The Debian package needs to be updated I'd gladly try to submit a fix, but there seems to be quite a lot of git repository layers in Launchpad and after spending a few time on it, I could not figure out where or how to do it.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39358 (Git)
Build type: Release
Branch: main
Hash: 735a07678d8682d5936906298ed61cb2efd0eb66
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * WikiPropertiesGenerator.FCMacro
  * fasteners 0.5.30
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

While at it, it might be worth updating the .deb package version from `0.22` to `1.1.0dev`

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-05:
@adrianinsaval

adrianinsaval, 2024-12-08:
should be fixed now

closed 2024-12-08 by adrianinsaval