FreeCAD/FreeCAD-Bundle #359: macos: boot error '`Failed to init from config`' ... '`PYTHONHOME is set to ... Setting this environment variable might cause Python to fail.`'

budiz1, 2024-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When i start the latest Freecad weekly build, i get an error that says (in the picture below).

I dont know if that is my computers problem or freecads as this error never showed up before.

[image]

Thank you for your feedback,
Budiz

### Full version info

```shell
I cant enter Freecad
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-23:
Moved to FreeCAD-Bundle

adrianinsaval, 2024-12-24:
Is this still relevant?

adrianinsaval, 2024-12-24:
> Moved to FreeCAD-Bundle

If we don't have confirmation that a stale issue is still relevant please don't bump it, otherwise the stale bot can't do it's job. Not to mention that it's not enabled in the FreeCAD-Bundle repo

closed 2025-02-03 by budiz1