FreeCAD/FreeCAD-Bundle #360: [Problem] Weekly Build for macOS is broken

ibrep, 2024-12-24:
[image]

ibrep, 2024-12-24:
[image]

JohnOCFII, 2024-12-24:
On macOS arm machines, there is an installation issue. Until the new installation setup is resolved, to test the weekly builds, the user needs to open a Terminal window and execute the following command:

`sudo xattr -r -d com.apple.quarantine /Applications/FreeCAD.app`

yannik-smedek, 2025-04-07:
> On macOS arm machines, there is an installation issue. Until the new installation setup is resolved, to test the weekly builds, the user needs to open a Terminal window and execute the following command:
> 
> `sudo xattr -r -d com.apple.quarantine /Applications/FreeCAD.app`

Executing this command is a security risk and not a proper solution in my opinion. Is there a reason why this hasn't been fixed yet?

oursland, 2025-04-08:
I have a PR that we'll get in this week that signs the weekly builds on the main FreeCAD repo.