FreeCAD/FreeCAD-Bundle #363: Cannot start FreeCAD 1.0.0 AppImage (x86_64)

nvdl, 2024-12-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Downloaded the latest version from https://www.freecad.org/downloads.php
URL: https://github.com/FreeCAD/FreeCAD/releases/download/1.0.0/FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage

But it aborts when trying to run it from the command line.

```
chmod +x FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage
./FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage
```

```
FreeCAD 1.0.0, Libs: 1.0.0R39109 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

AttributeError: module 'shibokensupport.signature.loader' has no attribute 'finish_import'
Fatal Python error: init_phase_2: could not initialize part 2
Python runtime state: initialized

Current thread 0x00007fa821d44840 (most recent call first):
  File "/home/nvd/.local/lib/python3.11/site-packages/shiboken2/__init__.py", line 30 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 940 in exec_module
  File "<frozen importlib._bootstrap>", line 690 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1147 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1176 in _find_and_load
  File "/tmp/.mount_FreeCAmiDdMj/usr/lib/python3.11/site-packages/PySide2/__init__.py", line 31 in _setupQtDirectories
  File "/tmp/.mount_FreeCAmiDdMj/usr/lib/python3.11/site-packages/PySide2/__init__.py", line 80 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 940 in exec_module
  File "<frozen importlib._bootstrap>", line 690 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1147 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1176 in _find_and_load
  File "/tmp/.mount_FreeCAmiDdMj/usr/Ext/PySide/__init__.py", line 2 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 940 in exec_module
  File "<frozen importlib._bootstrap>", line 690 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1147 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1176 in _find_and_load
  File "/tmp/.mount_FreeCAmiDdMj/usr/Mod/AddonManager/AddonManager.py", line 35 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 940 in exec_module
  File "<frozen importlib._bootstrap>", line 690 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1147 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1176 in _find_and_load
  File "/tmp/.mount_FreeCAmiDdMj/usr/Mod/AddonManager/InitGui.py", line 6 in <module>
  File "<string>", line 142 in RunInitGuiPy
  File "<string>", line 205 in InitApplications
  File "<string>", line 283 in <module>

Extension modules: FreeCAD, FreeCADGui, shiboken2.shiboken2, PySide2.QtCore (total: 4)
/tmp/.mount_FreeCAmiDdMj/AppRun: line 31: 113684 Aborted                 "${MAIN}" "$@"
```

OS: Debian GNU/Linux 12
Kernel: Linux 6.1.0-28-amd64

### Full version info

```shell
Cannot as it does not reach that point.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

oursland, 2025-01-01:
AppImage files contain all of the build artifacts along with the environment to run them.  FreeCAD is linked against the `libpython` within the environment.  The launch script sets up necessary environment variables to direct [`$PREFIX` and `$PYTHONHOME`](https://github.com/FreeCAD/FreeCAD-Bundle/blob/main/conda/linux/AppDir/AppRun#L3) and other variables to this environment.  In principle, FreeCAD should only use Python modules installed within this environment.

It is possible that other environment variables may somehow cause an incorrect discovery of Python modules.  Do you have any of [these environment variables](https://docs.python.org/3/using/cmdline.html#environment-variables) set in your host environment?

nvdl, 2025-01-04:
I have some environment variables set so that local Python scripts can be found.

But, the problem persists even after setting the "PATH" to a minimal value and clearing the other environment variables.

```
export PATH=/usr/local/bin:/usr/bin:/bin
export PYTHONPATH=""
export PYTHONPYCACHEPREFIX=""
```

Checking:
```
export | grep PYTHON
PYTHONPATH=''
PYTHONPYCACHEPREFIX=''
```

This only happens when there is "shiboken2" at:
`~/.local/lib/python3.11/site-packages/shiboken2`

I have moved away from packages in the "~/.local/lib" directory and have switched to a Python virtual environment.

But, it may affect someone who has stuff installed at "~/.local/lib".

The issue exists even if launching the "AppImage" from a Python virtual environment with "PATH" set to a minimal value (as above) and "PYTHONPATH" and "PYTHONPYCACHEPREFIX" set to nothing.

Removing the package from "~/.local/lib/python3.11/site-packages" fixes the issue.
`pip3 uninstall --break-system-packages shiboken2`

Interestingly, this issue does not exist if I switch to a Python virtual environment, install "shiboken2" and run the "AppImage".

```
# Switch to the virtual environment.
pip3 install shiboken2
./FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage
```