FreeCAD/FreeCAD-Bundle #375: Error importing numpy

Sil68, 2025-03-05:
Hi,

Just updated to most recent developer version of fc, and now -- upon launching fc -- the following error message is showing up:

```
09:42:36  Path WB not found
09:42:38  Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.
```

I'm using the python interpreter bundled with fc.

In the add-on manager I can find neither the path wb, nor the cam wb.

OS: macOS 15.4
Architecture: arm64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 72/108
Installed mods: 
  * MeshRemodel 1.10.37
  * ThreadProfile 1.92.0
  * Reporting
  * CommandPanel
  * BillOfMaterials 1.0.2
  * woodworking 0.21.2.33771.0.23.1
  * FreeGrid 2.2.0
  * ArchTextures
  * Defeaturing 1.2.3
  * Gridfinity 0.10.3
  * FEMbyGEN 2.3.0
  * TitleBlock 0.5.2.2
  * fasteners 0.5.37
  * Lithophane
  * lattice2 1.0.0
  * sheetmetal 0.7.21
  * CubeMenu
  * FEM_FrontISTR 0.2.0
  * MakerWorkbench 1.0.1
  * 3D_Printing_Tools
  * FrameForge 0.1.2
  * Curves 0.6.58
  * Assembly4 0.50.17

Sil68, 2025-03-07:
Launching fc from the command line results in the following output

```
(myusr) ➜  FreeCAD /Applications/Programmes/multimedia/FreeCAD.app/Contents/MacOS/FreeCAD --verbose
AR=arm64-apple-darwin20.0.0-ar
AS=arm64-apple-darwin20.0.0-as
BUILD=arm64-apple-darwin20.0.0
CC=arm64-apple-darwin20.0.0-clang
CC_FOR_BUILD=/usr/local/miniforge3/envs/myusr/bin/arm64-apple-darwin20.0.0-clang
CFLAGS=-ftree-vectorize -fPIC -fstack-protector-strong -O2 -pipe -isystem /usr/local/miniforge3/envs/myusr/include
CHECKSYMS=arm64-apple-darwin20.0.0-checksyms
CLANG=arm64-apple-darwin20.0.0-clang
CMAKE_ARGS=-DCMAKE_AR=/usr/local/miniforge3/envs/myusr/bin/arm64-apple-darwin20.0.0-ar -DCMAKE_CXX_COMPILER_AR=/usr/local/miniforge3/envs/myusr/bin/arm64-apple-darwin20.0.0-ar -DCMAKE_C_COMPILER_AR=/usr/local/miniforge3/envs/myusr/bin/arm64-apple-darwin20.0.0-ar -DCMAKE_RANLIB=/usr/local/miniforge3/envs/myusr/bin/arm64-apple-darwin20.0.0-ranlib -DCMAKE_CXX_COMPILER_RANLIB=/usr/local/miniforge3/envs/myusr/bin/arm64-apple-darwin20.0.0-ranlib -DCMAKE_C_COMPILER_RANLIB=/usr/local/miniforge3/envs/myusr/bin/arm64-apple-darwin20.0.0-ranlib -DCMAKE_LINKER=/usr/local/miniforge3/envs/myusr/bin/arm64-apple-darwin20.0.0-ld -DCMAKE_STRIP=/usr/local/miniforge3/envs/myusr/bin/arm64-apple-darwin20.0.0-strip -DCMAKE_INSTALL_NAME_TOOL=/usr/local/miniforge3/envs/myusr/bin/arm64-apple-darwin20.0.0-install_name_tool -DCMAKE_LIBTOOL=/usr/local/miniforge3/envs/myusr/bin/arm64-apple-darwin20.0.0-libtool -DCMAKE_BUILD_TYPE=Release -DCMAKE_OSX_SYSROOT=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk
CMAKE_PREFIX_PATH=:/usr/local/miniforge3/envs/myusr
COLORFGBG=0;15
COLORTERM=truecolor
COMMAND_MODE=unix2003
CONDA_BACKUP_HOST=MH-Mac-mini-05.local
CONDA_DEFAULT_ENV=myusr
CONDA_EXE=/usr/local/miniforge3/bin/conda
CONDA_PREFIX=/usr/local/miniforge3/envs/myusr
CONDA_PREFIX_1=/usr/local/miniforge3
CONDA_PROMPT_MODIFIER=(myusr)
CONDA_PYTHONBREAKPOINT=
CONDA_PYTHON_EXE=/usr/local/miniforge3/bin/python
CONDA_SHLVL=2
CONDA_TOOLCHAIN_BUILD=arm64-apple-darwin20.0.0
CONDA_TOOLCHAIN_HOST=arm64-apple-darwin20.0.0
CPP=arm64-apple-darwin20.0.0-clang-cpp
CPPFLAGS=-D_FORTIFY_SOURCE=2 -isystem /usr/local/miniforge3/envs/myusr/include
CPP_FOR_BUILD=/usr/local/miniforge3/envs/myusr/bin/arm64-apple-darwin20.0.0-clang-cpp
DEBUG_CFLAGS=-ftree-vectorize -fPIC -fstack-protector-strong -O2 -pipe -Og -g -Wall -Wextra -isystem /usr/local/miniforge3/envs/myusr/include
DISPLAY=/private/tmp/com.apple.launchd.SxAgB4k4tQ/org.xquartz:0
EDITOR=vi
EXINIT=set ai ic number ruler showmatch showmode ts=4 sw=4
FONTCONFIG_FILE=/etc/fonts/fonts.conf
FONTCONFIG_PATH=/etc/fonts
GIT_SSL_CAINFO=/Applications/Programmes/multimedia/FreeCAD.app/Contents/Resources/ssl/cacert.pem
GREP_COLORS=ms=01;31:mc=01;31:sl=:cx=:fn=35:ln=32:bn=32:se=36
GSETTINGS_SCHEMA_DIR=/usr/local/miniforge3/envs/myusr/share/glib-2.0/schemas
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=
HOME=/Users/myusr
HOST=arm64-apple-darwin20.0.0
INSTALL_NAME_TOOL=arm64-apple-darwin20.0.0-install_name_tool
ITERM_PROFILE=Default
ITERM_SESSION_ID=w0t1p0:73F06C8B-5B24-4838-B04A-962395354A52
JAVA_HOME=/Library/Java/JavaVirtualMachines/zulu-15.jdk/Contents/Home
LANG=UTF-8
LANGUAGE=en_GB
LC_TERMINAL=iTerm2
LC_TERMINAL_VERSION=3.5.12beta2
LD=arm64-apple-darwin20.0.0-ld
LDFLAGS=-Wl,-headerpad_max_install_names -Wl,-dead_strip_dylibs -Wl,-rpath,/usr/local/miniforge3/envs/myusr/lib -L/usr/local/miniforge3/envs/myusr/lib
LDFLAGS_LD=-headerpad_max_install_names -dead_strip_dylibs -rpath /usr/local/miniforge3/envs/myusr/lib -L/usr/local/miniforge3/envs/myusr/lib
LD_LIBRARY_PATH=/Applications/Programmes/multimedia/FreeCAD.app/Contents/Resources/lib
LESS=-R
LIBTOOL=arm64-apple-darwin20.0.0-libtool
LIPO=arm64-apple-darwin20.0.0-lipo
LOGNAME=myusr
LSCOLORS=Gxfxcxdxbxegedabagacad
LS_COLORS=di=1;36:ln=35:so=32:pi=33:ex=31:bd=34;46:cd=34;43:su=30;41:sg=30;46:tw=30;42:ow=30;43
LaunchInstanceID=FF813EF3-EFAE-44E1-9FF3-CE2161774453
MAMBA_ROOT_PREFIX=/usr/local/miniforge3
MANPATH=/opt/homebrew/man:/usr/local/man:
MESON_ARGS=-Dbuildtype=release
NM=arm64-apple-darwin20.0.0-nm
NMEDIT=arm64-apple-darwin20.0.0-nmedit
OBJC=arm64-apple-darwin20.0.0-clang
OBJC_FOR_BUILD=/usr/local/miniforge3/envs/myusr/bin/arm64-apple-darwin20.0.0-clang
OLDPWD=/Users/myusr/Library
[... 39 lines cut ...]
__CFBundleIdentifier=com.googlecode.iterm2
__CF_USER_TEXT_ENCODING=0x0:0:2
ac_cv_func_malloc_0_nonnull=yes
ac_cv_func_realloc_0_nonnull=yes
build_alias=arm64-apple-darwin20.0.0
host_alias=arm64-apple-darwin20.0.0
Running: /Applications/Programmes/multimedia/FreeCAD.app/Contents/Resources/bin/freecad
Error: Failed to open library "/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib"! Error: dlopen(/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib, 0x0005): tried: '/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/System/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file, not in dyld cache)!
FreeCAD 1.1.0, Libs: 1.1.0devR40504 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

addFC - additional tools for FreeCAD
FEMbyGEN Workbench loaded
[... cut ...]
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there." occurred in /Users/myusr/Library/Application Support/FreeCAD/Mod/addFC/./InitGui.py
Please look into the log file for further information
Path WB not found
libpng warning: iCCP: duplicate
Wizard shaft module cannot be loaded
```

I wiped fc completely, re-installed it from scratch after downloading the package from github again, still the very same result, regardless if just plain vanilla fc, or some additional wb's installed

Sil68, 2025-03-07:
Had to install the 3dconnexion driver in order to get rid of this part of the error message, despite having no 3dconnexion devices at all

Sil68, 2025-03-07:
How can I resolve this one

```
During initialization the error "Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there." occurred in /Users/myusr/Library/Application Support/FreeCAD/Mod/addFC/./InitGui.py
Please look into the log file for further information
```

Also occuring eg. with the asm4 wb

oursland, 2025-03-07:
`numpy` is included in the releases, so this error does not make much sense to me.

I installed the most recent version of `addFC` and encounter no such error.  This may be an issue that should be raised with the `addFC` workbench repository.

Sil68, 2025-03-08:
removed addfc, now the very same issue with asm4

Sil68, 2025-03-10:
just wiped my fc install -- programme, config, cache, preferences -- , downloaded & installed most recent weekly build, and launched this vanilla install of fc.

received some error message

```
qt.qpa.fonts) Populating font family aliases took 47 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost.
Path WB not found
libpng warning: iCCP: duplicate
```

then switched to the cam wb (some sources on the net suggersted that the missing path wb has been replaced by cam wb) bundled with fc, and received even more error messages

```
Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.
Traceback (most recent call last):
  File "/Applications/Programmes/multimedia/FreeCAD.app/Contents/Resources/Mod/CAM/InitGui.py", line 82, in Initialize
    import PathCommands
  File "/Applications/Programmes/multimedia/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/Programmes/multimedia/FreeCAD.app/Contents/Resources/Mod/CAM/PathCommands.py", line 28, in <module>
    from PathScripts.PathUtils import loopdetect
  File "/Applications/Programmes/multimedia/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/Programmes/multimedia/FreeCAD.app/Contents/Resources/Mod/CAM/PathScripts/PathUtils.py", line 30, in <module>
    from numpy import linspace
  File "/Applications/Programmes/multimedia/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/Programmes/multimedia/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/numpy/__init__.py", line 119, in <module>
    raise ImportError(msg) from e
```

oursland, 2025-03-10:
That is expected as the python interpreter is not expected to be executed directly.  The launcher sets some environment variables necessary for Python to find the paths to the installed modules: https://github.com/FreeCAD/FreeCAD-Bundle/blob/main/conda/osx/launcher/FreeCAD.cpp

momja, 2025-05-05:
I also receive the same error on the latest weekly build while trying to open Assembly4 workbench. MacOS 15.3.

```
Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.
18:18:19  Traceback (most recent call last):
  File "/Users/{redacted}/Library/Application Support/FreeCAD/Mod/Assembly4/./InitGui.py", line 151, in Initialize
    import AnimationLib        # creates an LCS in assembly and attaches it to an LCS relative to an external file
    ^^^^^^^^^^^^^^^^^^^
  File "/Users/{redacted}/Library/Application Support/FreeCAD/Mod/Assembly4/./AnimationLib.py", line 11, in <module>
    import os, numpy
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/numpy/__init__.py", line 119, in <module>
    raise ImportError(msg) from e
```

luzpaz, 2025-05-10:
Is this bug macOS specific ?

oursland, 2025-05-11:
Yes.  It's an upstream issue with [`numpy-feedstock`](https://github.com/conda-forge/numpy-feedstock/issues/347) when installed with `pixi` and run on macOS 15.4.  I'm going to submit a PR that *may* fix the issue in the next weekly build.

closed 2025-07-16 by hyarion