FreeCAD/FreeCAD-Bundle #376: Weekly build 40655 does not start up on macOS x86

depthoffocus, 2025-03-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Not sure if this should be reported here or in the FreeCAD-Bundle repo.

The current weekly build **40655** does not properly start up on macOS x86. 

I had removed the quarantine as normal with 

```sh
xattr -r -d com.apple.quarantine "/Applications/FreeCAD.app"
```

FreeCAD then verifies as normal on first launch but silently quits.

On subsequent launches the splash screen appears briefly then disappears. There is no crash. The app appears again to be silently quitting.

Starting the app in safe mode with:

```sh
open /Applications/FreeCAD.app --args --safe-mode 
```

does not help. 

(That is the correct way to start safe mode, verified with other builds.)

There is no console output at that point.

Information that might be useful:

MacBook Pro (13-inch, 2019, touchbar) -- `MacBookPro15,2`
macOS Sequoia 15.3.1 (24D70)

(I am unable to get the FreeCAD build info from the about panel because things do not get that far).

### Full version info

```shell
--
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-11:
@adrianinsaval @oursland FYI

modevpro, 2025-03-11:
Weekly Build 40655 ARM version also does not open on Macbook Pro M1 macOS Sequoia 15.3.1.
Shows splash screen and crashes

sylanc3, 2025-03-12:
I can confirm here same behaviour for ARM or x86 Version on a MacBook Pro M1.

oursland, 2025-03-13:
I can confirm this as well.  I do have the 3Dconnexion driver installed, so that is not the source of the issue.

JohnOCFII, 2025-03-13:
Same behavior here -- the weekly (40655) fails to start, but self-built version (using Pixi build process) works fine on my M1 MacBook Pro.

fredjohnston, 2025-03-14:
> I can confirm here same behaviour for ARM or x86 Version on a MacBook Pro M1.

And same on MacbookPro M4

maxwxyz, 2025-03-15:
They are not updated for some days, @oursland is currently migrating the workflow to the main repo

oursland, 2025-03-15:
The build fails because these builds utilize Qt5 and C++17, but the main branch has moved to Qt6 and C++20.  It has been observed that the migration from Qt5 to Qt6 has resulted in some behavior changes and crashes, it was considered desirable to have a way to return to a previous working release.

The plan for upgrading the build is as follows (discussed at the March 2 FreeCAD Developer Meeting): 
1. Create weekly archives of builds so that if a given build is defective, the previous build can be re-installed
    This is to be done by creating tag in the main repo, performing the build, and creating a "pre-release" release with build artifacts
2. Perform the upgrade to Qt6 and C++20

Initially this plan was intended to take place before the `FreeCAD-Bundle` build broke, but recent PRs have made it impossible to build the weekly builds.  Furthermore, the most recent successful weekly build appears to be defective, and without the archive there is no way to return to a previously functioning build.

As the weekly builds are effectively broken and there are no archives of prior working builds, it may be beneficial to simply migrate the current weekly build system to Qt6 and C++20 and accept that there are some regressions, which is still better than a completely non-functional build.

maxwxyz, 2025-03-15:
Yes this seams sensible. The earlier the "buggy" builds are published, the more Qt6 / C++20 bugs can be found (and hopefully fixed) prior to the next release.

Andor, 2025-03-18:
Guys, could you please re-release older working build? It is still better than having only non-working build available.

pixelwaster, 2025-03-22:
> The build fails because these builds utilize Qt5 and C++17, but the main branch has moved to Qt6 and C++20. It has been observed that the migration from Qt5 to Qt6 has resulted in some behavior changes and crashes, it was considered desirable to have a way to return to a previous working release.
> 
> 
> As the weekly builds are effectively broken and there are no archives of prior working builds, it may be beneficial to simply migrate the current weekly build system to Qt6 and C++20 and accept that there are some regressions, which is still better than a completely non-functional build.

I have **FreeCAD_weekly-builds-37645-conda-Windows-x86_64-py311** thru to **FreeCAD_weekly-builds-40421-conda-Windows-x86_64-py311**  if you would like me to upload them somewhere. I am also willing to bet there are other users out there will some archives. I know of one but he is also on Windows. 

Posting in the forum that you are looking for archives, I found this topic by accident. Another idea is to ask some of the well known YouTube channels if they would post a 2 minute video asking for archives.

ademlabs, 2025-03-22:
I have FreeCAD_weekly-builds-40006-conda-macOS-arm64-py311.dmg. I had to go back to that version after trying the recent broken build.

Andor, 2025-03-22:
I build relatively quickly with `pixi`. Here is the instruction: https://freecad.github.io/DevelopersHandbook/gettingstarted/#pixi

psommerfeld, 2025-03-22:
I have FreeCAD_weekly-builds-40176-conda-macOS-arm64-py311.dmg available. It's the last functional MacOS build I think.

sylanc3, 2025-03-22:
> I build relatively quickly with `pixi`. Here is the instruction: https://freecad.github.io/DevelopersHandbook/gettingstarted/#pixi

Hello @Andor , thanks for the tip with pixi. Was quite easy to build. I can also execute the binary. But how to create an App Bundle? (I'm on a mac, no developer experience)
BR Flow

Andor, 2025-03-22:
@sylanc3 I will just run it with `pixi run freecad` until the official build came out. Works for me.

hyarion, 2025-03-22:
Pixi does not yet build bundles. As Andro mentions running it like that works, but you can probably also navigate to build/bin with finder and double click freecad in there.

FC-FBXL5, 2025-03-23:
Weekly build 40928 also fails.

depthoffocus, 2025-03-23:
> Weekly build 40928 also fails.

Indeed it does, but it's failing in a different way to my original report. This time, I get an actual crash report on startup a few seconds after the initial verification, rather than a silent (when launched with a double-click at least) failure to start.

It's a SIGABRT, inside Qt.

So I think this may be a different issue. (The pixi build I tried yesterday did not crash like this though.)

Relevant bits of the crash log:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [67255]
Path:                  /Applications/FreeCAD.app/Contents/MacOS/FreeCAD
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-03-23 17:45:42.6455 +0000
OS Version:            macOS 15.3.2 (24D81)
Report Version:        12
Bridge OS Version:     9.3 (22P3060)
Anonymous UUID:        5FF5BD02-78A8-4188-0EAB-EF410C648824

Sleep/Wake UUID:       2B7C8B9C-9821-4A33-9628-D4B6919BCEAC

Time Awake Since Boot: 860000 seconds
Time Since Wake:       337200 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   freecad [67255]

Application Specific Information:
abort() called

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff804ecac52 __pthread_kill + 10
1   libsystem_pthread.dylib       	    0x7ff804f04f85 pthread_kill + 262
2   libsystem_c.dylib             	    0x7ff804e25b19 abort + 126
3   libFreeCADGui.dylib           	       0x10cc5ab75 messageHandler(QtMsgType, QMessageLogContext const&, QString const&) + 517
4   libQt6Core.6.7.3.dylib        	       0x10c42fedb qt_message_print(QtMsgType, QMessageLogContext const&, QString const&) + 251
5   libQt6Core.6.7.3.dylib        	       0x10c42b8ef qt_message(QtMsgType, QMessageLogContext const&, char const*, __va_list_tag*) + 63
6   libQt6Core.6.7.3.dylib        	       0x10c4a91f6 QMessageLogger::fatal(char const*, ...) const + 124
7   libQt6Gui.6.7.3.dylib         	       0x10faf419d init_platform(QString const&, QString const&, QString const&, int&, char**) + 8301
8   libQt6Gui.6.7.3.dylib         	       0x10faf08e3 QGuiApplicationPrivate::createPlatformIntegration() + 1715
9   libQt6Gui.6.7.3.dylib         	       0x10faf444b QGuiApplicationPrivate::createEventDispatcher() + 27
10  libQt6Core.6.7.3.dylib        	       0x10c3c8014 QCoreApplicationPrivate::init() + 1716
11  libQt6Gui.6.7.3.dylib         	       0x10faecc19 QGuiApplicationPrivate::init() + 57
12  libQt6Widgets.6.7.3.dylib     	       0x10f395e33 QApplicationPrivate::init() + 51
13  libFreeCADGui.dylib           	       0x10cd22717 Gui::GUIApplicationNativeEventAware::GUIApplicationNativeEventAware(int&, char**) + 23
14  libFreeCADGui.dylib           	       0x10cd1f853 Gui::GUIApplication::GUIApplication(int&, char**) + 35
15  libFreeCADGui.dylib           	       0x10cd205e5 Gui::GUISingleApplication::GUISingleApplication(int&, char**) + 37
16  libFreeCADGui.dylib           	       0x10cc5b17c Gui::Application::runApplication() + 140
17  freecad                       	       0x10a89a7c8 main + 4104
18  dyld                          	    0x7ff804b762cd start + 1805

```

Relevant message if I start it from the command line:

```
mike@Michaels-MacBook-Pro ~ % /Applications/FreeCAD.app/Contents/MacOS/FreeCAD 
[...]
Running: /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Error: Failed to open library "/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib"! Error: dlopen(/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib, 0x0005): tried: '/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/System/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file, not in dyld cache)!
FreeCAD 1.1.0, Libs: 1.1.0devR40928 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Detected locale "C" with character encoding "US-ASCII", which is not UTF-8.
Qt depends on a UTF-8 locale, and has switched to "UTF-8" instead.
If this causes problems, reconfigure your locale. See the locale(1) manual
for more information.
(qt.qpa.plugin) Could not find the Qt platform plugin "cocoa" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

zsh: abort      /Applications/FreeCAD.app/Contents/MacOS/FreeCAD
```

fester225, 2025-03-23:
I just got the same Problem Report.

rtm7777, 2025-03-24:
> I have FreeCAD_weekly-builds-40176-conda-macOS-arm64-py311.dmg available. It's the last functional MacOS build I think.

Hello, can I get it somehow please? In my project I'm heavily relying on 1.1 features and can't use 1.0 anymore, for 2 weeks I'm just waiting 🙈

hyarion, 2025-03-24:
> > I have FreeCAD_weekly-builds-40176-conda-macOS-arm64-py311.dmg available. It's the last functional MacOS build I think.
> 
> Hello, can I get it somehow please? In my project I'm heavily relying on 1.1 features and can't use 1.0 anymore, for 2 weeks I'm just waiting 🙈

You have an unofficial mac build of 40913 that you could try here (mac+linux should work but windows is still failing): https://github.com/oursland/FreeCAD/releases/tag/weekly-2025.03.24
oursland is still working on getting everything up running, hopefully this will be fixed in the next couple of days.

modevpro, 2025-03-24:
> You have an unofficial mac build of 40913 that you could try here (mac+linux should work but windows is still failing): https://github.com/oursland/FreeCAD/releases/tag/weekly-2025.03.24
oursland is still working on getting everything up running, hopefully this will be fixed in the next couple of days.

Hello, this build [FreeCAD_weekly-2025.03.24-macOS-arm64-py313.dmg](https://github.com/oursland/FreeCAD/releases/download/weekly-2025.03.24/FreeCAD_weekly-2025.03.24-macOS-arm64-py313.dmg) freezes after launch on Macbook Pro M1.

hyarion, 2025-03-24:
> Hello, this build [FreeCAD_weekly-2025.03.24-macOS-arm64-py313.dmg](https://github.com/oursland/FreeCAD/releases/download/weekly-2025.03.24/FreeCAD_weekly-2025.03.24-macOS-arm64-py313.dmg) freezes after launch on Macbook Pro M1.

It works on my 14-inch, 2021 Apple M1 Pro running macOS 14.5 (downloaded and tested it again just to be sure)

hyarion, 2025-03-24:
@modevpro how about this: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/devel-qt5 ?

modevpro, 2025-03-24:
> [@modevpro](https://github.com/modevpro) how about this: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/devel-qt5 ?

[FreeCAD_devel-qt5-40655-conda-macOS-arm64-py311.dmg](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/devel-qt5/FreeCAD_devel-qt5-40655-conda-macOS-arm64-py311.dmg) this build crashes on Macbook Pro M1 macOS Sequoia 15.3.2

```
➜  ~ /Applications/FreeCAD.app/Contents/MacOS/FreeCAD
[...]
Running: /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Error: Failed to open library "/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib"! Error: dlopen(/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib, 0x0005): tried: '/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/System/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file, not in dyld cache)!
FreeCAD 1.1.0, Libs: 1.1.0devR40655 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Error in InitApplications embedded null byte
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 707, in <module>
  File "<string>", line 98, in InitApplications
  File "<frozen posixpath>", line 416, in realpath
  File "<frozen posixpath>", line 451, in _joinrealpath
ValueError: embedded null byte
--------------------------------------------------------------------------------
(qt.qpa.fonts) Populating font family aliases took 54 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
Traceback (most recent call last):
  File "<string>", line 303, in <module>
  File "<string>", line 133, in InitApplications
<class 'AttributeError'>: module 'FreeCAD' has no attribute '__ModDirs__'
Error in FreeCADGuiInit.py: module 'FreeCAD' has no attribute '__ModDirs__'
2025-03-24 15:30:30.909 freecad[15632:220087] +[IMKClient subclass]: chose IMKClient_Modern
2025-03-24 15:30:30.909 freecad[15632:220087] +[IMKInputSession subclass]: chose IMKInputSession_Modern
```

fester225, 2025-03-24:
I just got FreeCAD_weekly-2025.03.24-macOS-arm64-py313.dmg to open on my Mac Mini (ARM64)(ver15.3.2) by using;

sudo xattr -rd com.apple.quarantine /Applications/FreeCAD.app

depthoffocus, 2025-03-24:
> Hello, this build [FreeCAD_weekly-2025.03.24-macOS-arm64-py313.dmg](https://github.com/oursland/FreeCAD/releases/download/weekly-2025.03.24/FreeCAD_weekly-2025.03.24-macOS-arm64-py313.dmg) freezes [...]

It freezes for me, too (x86 mac). Including in safe mode. Shortly after startup, regardless of what I do.

depthoffocus, 2025-03-25:
Success! Build 40971 *does* successfully start for me.

hyarion, 2025-03-25:
I'm closing this issue as OP has verified that the new version works.