FreeCAD/FreeCAD-Bundle #382: MacOS weekly build fails to open on MacBook Air

KeithSloan, 2025-03-20:
Cannot open weekly build 

 [FreeCAD_weekly-builds-40655-conda-macOS-arm64-py311.dmg](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-40655-conda-macOS-arm64-py311.dmg)

MacBook Air
Sequoia 15.3.2

rtm7777, 2025-03-20:
https://github.com/FreeCAD/FreeCAD-Bundle/issues/376

KeithSloan, 2025-03-20:
Don't think #376 is the same it referees tio an arch x86 build not arch64, also the problem is MacOS veriffies the file and will not open it.

rtm7777, 2025-03-21:
I have the same on arm64 M2 mac

KeithSloan, 2025-03-23:
I just tried a download again today and still a problem.

adrianinsaval, 2025-03-25:
please try the new build

KeithSloan, 2025-03-25:
> please try the new build

Will do.

Also I noticed that one of the AstoCAD FAQ's had the same issue and it was to do with Apple signing and there was a work around, which I thought was similar to something I tried from a suggestion I found on the web somewhere

closed 2025-03-25 by hyarion