FreeCAD/FreeCAD-Bundle #383: Latest Appimage Does Not Work In Debian 12

KitchM, 2025-03-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Downloaded FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage.  Used AppimageLauncher and I tried it by integrating it and also by running it once.  It does not start up.

In the first case, it returned an error: "failed to register appimage in system via libappimage"

### Full version info

```shell
FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage
```

### Subproject(s) affected?

None

### Anything else?

Debian 12 Bookworm and Xfce 4.18.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-03-20:
https://github.com/TheAssassin/AppImageLauncher/issues/410

KitchM, 2025-03-20:
Sorry, missed that one.  I'll check into it.

Thanks.

closed 2025-03-20 by KitchM