FreeCAD/FreeCAD-Bundle #384: AppImage Version Does Not Start

KitchM, 2025-03-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Neither Integrate and Run or Run Once allows FreeCAD to start when using AppImageLauncher.  Evidently there is a package creation error as detailed here:
[https://github.com/TheAssassin/AppImageLauncher/issues/410](url)

### Full version info

```shell
FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage
```

### Subproject(s) affected?

None

### Anything else?

Debian 12 with Xfce 4.18

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-21:
@adrianinsaval FYI

leoheck, 2025-03-21:
It does not work with AppImageLauncher anymore and this sucks so bad.
They should fix this because the AppImageLauncher is a must.

@KitchM, can you confirm that you can at least launch the FreeCad AppImage without the AppImageLauncher?

michaelnu, 2025-03-21:
For me the appimage 1.0 is working directly with Debian 12 (with KDE).
The only thing what you have to do is to give the file executable rights.
If you start the appimage from console, what errormessage do you see? Without console?
Do you have a special graphic card?

CIKoolK, 2025-03-21:
I was not aware of _AppImageLauncher'_s existence until this issue arose late last year when FC updated their _.AppImage_ bundles to use the latest _.AppImage_ compression algorithm. 

I had never felt a need for such a thing either.

For years I have been extracting _.AppImage_ packages to dedicated folders and launching the _AppRun_ script or executable contained therein; it makes the application load faster and run smoother also. I've even created a script to do the extraction, rename the default _squashfs-root_ folder, and if desired create a _desktop launcher_.

The current FC .AppImage packages are working with no issues.

_Mint v22.1/MATÉ_

leoheck, 2025-03-21:
@CIKoolK your comment is funny. Are you using AppImageLauncher? If not what your reply has to do with this issue report?

KitchM, 2025-03-21:
I tried to run the program, but appimagelauncherd is running in the background.  So I used the task manager to shut off that item.  However, every time I wish to start the program the launcher interface comes up anyway.

All I know is that I had it running in the past, and I believe I did not have appimagelauncher running then.

CIKoolK, 2025-03-21:
> [@CIKoolK](https://github.com/CIKoolK) your comment is funny. Are you using AppImageLauncher? If not what your reply has to do with this issue report?

The "relevance" is that the _FC_ and _AstoCAD_  .AppImage bundles are not at fault here--_AppImageLauncher_ is the issue--it does not support the current, most efficient compression algorithm used by _.AppImage_.

For those having _AppImageLauncher_ installed and having difficulty launching _.AppImage_ files, it is most likely _AppImageLauncher_ commandeered the _.AppImage_ "extension" in _$HOME/,config/mimeapps.list_ or elsewhere...

CIKoolK, 2025-03-21:
> I tried to run the program, but appimagelauncherd is running in the background. So I used the task manager to shut off that item. However, every time I wish to start the program the launcher [...]

_AppImageLauncher_ has likely commandeered the _.AppImage_ "extension" in _$HOME/,config/mimeapps.list_ or elsewhere.

VM4Dim, 2025-03-21:
Changing extension of file from `.AppImage` to another should help.

CIKoolK, 2025-03-22:
> Changing extension of file from `.AppImage` to another should help.

I do not believe this will assist, Linux does not set great "store" in file extensions--(oversimplified!) it will examine the file, see that it is of type "ELF" (Executable Link Format) in the 1st few bytes read the ELF header and hand it off to whatever application is registered to deal with it.

I just conducted an exercise to "see what I could see":

I 1st created a _Timeshift_ snapshot, so if "it" hit the fan I could "get back to sanity";

Next I installed "_appimagelauncher_3.0.0-alpha-4-gha253.36951ec_amd64.deb_", the latest version I found on _Github_;

Sure enough _AppImageLauncher_ now believed itself to be the "owner" of _.AppImage_ packages;

I renamed "_FreeCAD-40655.AppImage_" to "_FreeCAD-40655.xxx_", and _AppImageLauncher_ gladly popped up when I attempted to launch it;

I poked about looking for how AppImageLauncher had inserted itself into my system but nothing was obvious, it was not through the mimeapp,list file as I had speculated earlier--and as I really don't care I'm going to bed.

The good news is that "dpkg -r" uninstalled _AppImageLauncher_ and it's intrusions--so restoring the snapshot was not needed.

BTW that latest version of _AppImageLauncher_ did launch the FreeCAD 40655 build. I had noted it was released in February, it appears it was updated to use the new _.AppImage_ compression algorithm.

NOTE: **Uninstalling _AppImageLauncher_ also delected the FreeCAD-40655.xxx file I had launched with it!**

CIKoolK, 2025-03-22:
Insomnia now has me:

I will stand by my assertion above that AppImageLauncher is uneeded--do this:

- Copy the target .AppImage package (let's say "MyAppImage,AppImage") to a desired location;
- At a CLI "in that folder" execute `MyAppImage,AppImage --appimage-extract`
- The _.appimage_ contents will be extracted to a folder named _squashfs-root_;
- Rename that folder as you like;
- Execute the _AppRun_ bash script contained therein, create a desktop launcher. menu entries. whatever as you like;

You will find the application launches faster and runs more smoothly...

CIKoolK, 2025-03-22:
Well, I ended up restoring from the "snapshot". Something I could not quite "place" did not feel right and I did not wish to persist it...

_**"If you do what your gut tells you to do you might be wrong--if you don't you probably will be wrong!"**_
-me- ca. 1992

CIKoolK, 2025-03-22:
Woke up ambitious, booted _Mate v22.1/Cinnamon_ from an alternate drive and loaded _AppImageLauncher_ (_AIL_)--doesn't matter on this drive as I do not like _Cinnamon_.

I found that _AIL_ is not listed in the system services, will have to do more homework on that--IDK.

Also found it took 10-15 s to load _FC b40655_ via _AIL_, vs. 2-3 s from an extracted archive folder.

VM4Dim, 2025-03-22:
@CIKoolK 
You are right, it works not through the file extension.

It used binfmt and "magic symbols" `AI` with offset 8 bytes.

/opt/appimagelauncher.AppDir/usr/lib/binfmt.d/appimagelauncher.conf ( from: appimagelauncher_3.0.0-alpha-4 *.deb )

```
# Launch AppImages with AppImageLauncher
# we use the F (fix binary) mode to make sure the interpreter is available in all namespaces/cgroups/changerooots
# without having to have the binary in a specific place
:appimage-type1:M:8:AI\x01::/opt/appimagelauncher.AppDir//usr/lib/x86_64-linux-gnu/appimagelauncher/binfmt-interpreter:F
:appimage-type2:M:8:AI\x02::/opt/appimagelauncher.AppDir//usr/lib/x86_64-linux-gnu/appimagelauncher/binfmt-interpreter:F
```

CIKoolK, 2025-03-22:
> [@CIKoolK](https://github.com/CIKoolK) You are right, it works not through the file extension.
>
> It used binfmt and "magic symbols" `AI` with offset 8 bytes. [...]

That _Linux_ does not slavishly consider file extensions significant indicators of anything is one of it's greatest, and often overlooked advantages. It contributes a lot to _Linux_' overall security superiority!

KitchM, 2025-03-26:
To be clear, I used sudo apt remove appimagelauncher and then tried to execute FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage and it started right up.

CIKoolK, 2025-03-26:
That's the way to do it, AppImageLauncher embeds itself into the o/s somewhere (I have not yet sorted out where) and intercepts the launching of .AppImage packages--I would not have it just for that. 

Complete "surgical" removal is needed to get it all out...

KitchM, 2025-03-27:
So, there is a compatibility issue; but on whose side?

CIKoolK, 2025-03-27:
It appears a recent release of AppImageLauncher has been updated to recognize the new compression algorithm being used by the AppImage packaging.  It was an AppImageLauncher problem...

JohnOCFII, 2025-03-27:
> Sure, so as I said. The AppImage _is working_. What is not working is the compatibility with the AppImageLauncher.

Isn't this the same issue that was acknowledged back in November?  https://github.com/FreeCAD/FreeCAD-Bundle/issues/331 and https://github.com/TheAssassin/AppImageLauncher/issues/602

leoheck, 2025-03-27:
> So, there is a compatibility issue; but on whose side?

Freecad, of course. It used to work just fine.

CIKoolK, 2025-03-27:
> > So, there is a compatibility issue; but on whose side?
> 
> Freecad, of course. It used to work just fine.

NO, FreeCAD switched to using the new .AppImage compression algorithm--AppImageLauncher did not until it's latest version...

closed 2025-03-31 by hyarion