FreeCAD/FreeCAD-Bundle #387: Appimage won't start anymore.

oficinerobotica, 2025-03-23:
When I try to run FreeCAD_weekly-builds-40928-conda-Linux-x86_64-py311.AppImage I get:

```FreeCAD 1.1.0, Libs: 1.1.0devR40928 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

NOTICE: Addon "/home/flo/.local/share/FreeCAD/Mod/sheetmetal.backup1729482375.2960367" disabled by presence of ADDON_DISABLED stopfile
(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
(qt.qpa.plugin) Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

/tmp/.mount_FreeCAniJMcm/AppRun: line 31: 13910 Aborted                 (core dumped) "${MAIN}" "$@"
```

This issue was not present in FreeCAD_devel-qt5-40655-conda-Linux-x86_64-py311.AppImage

huguesdpdn-aerospace, 2025-03-23:
Hello,

same here for the last version. Got:
```
FreeCAD 1.1.0, Libs: 1.1.0devR40928 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

(qt.qpa.plugin) Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

/tmp/.mount_FreeCApLDCBL/AppRun : ligne 31 : 21315 Abandon                 (core dumped) "${MAIN}" "$@"
```

I removed QT, reinstall it, and same.

Old version of weekly downloaded FreeCAD works (I have a ~40000)

CIKoolK, 2025-03-23:
I also...

huguesdpdn-aerospace, 2025-03-23:
Related discussion : [https://forum.freecad.org/viewtopic.php?p=817872](https://forum.freecad.org/viewtopic.php?p=817872)

CIKoolK, 2025-03-23:
QT sucks--it is designed with zero backward compatibility, each new version breaks everything in sight, and installing it requires a Phd. in Advanced Geekology.

I went through this same B.S. with V5 breaking 4 or 5 applications.

VM4Dim, 2025-03-23:
Unpack and add file: `squashfs-root/usr/bin/qt6.conf` in unpacked AppImage.

```
[Paths]
Prefix    = ..
Libraries = lib
Plugins   = lib/qt6/plugins
```

thomasgi1, 2025-03-23:
> Unpack and add file: `squashfs-root/usr/bin/qt6.conf` in unpacked AppImage.
> 
> ```
> [Paths]
> Prefix    = ..
> Libraries = lib
> Plugins   = lib/qt6/plugins
> ```

On MacOS you can directly put that 'qt6.conf' file into 'FreeCAD.app/Contents/Resources/bin/' folder.

huguesdpdn-aerospace, 2025-03-23:
Indeed, same error. Trying to find the root cause.

- If I find it, I will update my PR.
- If not, please do not hesitate to help

CIKoolK, 2025-03-23:
> Indeed, same error. Trying to find the root cause.
> 
>     * If I find it, I will update my PR.
> 
>     * If not, please do not hesitate to help
>     * 

It (FC) seems to be performing correctly, however as a retired ME/ now MSLA hobbyist my needs are limited (simple Sketches, Part Design, Part, Draft, Mesh)...

CIKoolK, 2025-03-23:
FWIW, my STL export macro still works as it's supposed to:

```
# STL2Slicer.FCMacro - v1.2 - Septemner 2024 by paladin@paladinmicro.com
#
import FreeCAD
import Mesh
import subprocess
from PySide import QtGui
from pathlib import Path

macroName = "STL2Slicer"
# !!!
# !!! - user needs to set FQFN of slicer application, and mesh tolerance if desired--moved to top of code to be more visible
slicerFQFN = "/media/Applications/Chitubox_Basic231/CHITUBOX_Basic.sh" 
# slicerFQFN = "/media/Applications/CHITUBOX-V195/chitubox.sh"
# Mesh output tolerance (mm), overrides the setting in Preferences if smaller
meshTol = 0.010            

# some globals
exportList = ""              # Holds a list of objects to export

# Create a QMessageBox object
msgBox = QtGui.QMessageBox
  
# wrap it in a function that returns the button pressed
def progressDlg(Title, Msg, Btns, Type):
  dlg = msgBox()
  dlg.setWindowTitle(Title)
  dlg.setText(Msg)
  dlg.setStandardButtons(Btns)   
  dlg.setIcon(Type)
    # execute and return button
  return dlg.exec_()
  
# show user cancel message..  
def showUserCancel():
  Msg = "Cancelled by user..."
  Btns = msgBox.Ok
  usrSel = progressDlg(macroName, Msg, Btns, msgBox.Information)   
    
# add selected objects to output collection (objsSel[], build a list of their lalels (exportList))
# except "Origin" as it is not a "real" object
def getSelectedOjects(Objs):
  # Use the global list
  global exportList
  objsSel = []
  exportList = ""
  for obj in Objs:
 #   print(obj.Name)
    if obj.Name != "Origin":
      objsSel.append(obj) 
      exportList = exportList + "    " + obj.Label + "\n" 
  return objsSel
             
#             
# Try to replace .FCStd in document FQFN with .stl to form output filename
# If this fails the selected tab is not an FC project --STOP!!! 
try: 

  outFQFN = FreeCAD.ActiveDocument.FileName.replace(".FCStd", ".stl")
  
#
# We got here because there is no object document open    
except:
  # tell user, then stop
  Msg = "Please Open at Least One Saved Project and Try Again!"
  msgType = msgBox.Warning
  Btns = msgBox.Ok
  # show it...
  usrSel = progressDlg(macroName, Msg, Btns, msgType)

# Otherwise we can do this...    
else: 
  # Get selected objects, except "Origin"
  objsSelected = getSelectedOjects(Gui.Selection.getSelection())
  
  # Set a default
  usrSel = msgBox.Ok
  
  #Did we get some?
  if len(objsSelected) != 0:  

    # Does the outFQFN already exist?
    fileExists = Path(outFQFN).is_file()

    # tell user, let them cancel
    Msg = 'Preparing to Export:\n' + outFQFN
    msgType = msgBox.Information
    Btns = msgBox.Ok | msgBox.Cancel

    # If the file exists alert the user
    if fileExists:
      Msg = "!!!ALERT--THIS IS AN EXISTING FILE!!!\nDo you wish to overwrite?\n\n" + Msg + "\n"
      msgType = msgBox.Warning
      Btns = msgBox.No | msgBox.Yes

    # Get user input
      usrSel = progressDlg(macroName, Msg, Btns, msgType)
      
  # if user pressed [OK] (export) or [Yes] (overwrite) or not asked as there is nothing to export then try it...
  if (usrSel == msgBox.Ok) or (usrSel == msgBox.Yes):  

    # check that there's something to export, advise if not...
    # objsSelected = []        # for testing
    if len(objsSelected) != 0:  
                              
      # Tell user what's up
      Msg = "Exporting:\n\n" + exportList +"\n @" + str(meshTol*1000) + " µm deviation"
      msgType = msgBox.Information
      Btns = msgBox.Cancel | msgBox.Ok
      usrSel = progressDlg(macroName, Msg, Btns, msgType)      
     
      # User says do it!     
      if usrSel == msgBox.Ok:
        
        # export visible objects to the .stl file (outFQFN), uses tolerance set above
        Mesh.export(objsSelected, outFQFN, meshTol)
        
       # open the slicer
        rslt = subprocess.Popen([slicerFQFN, outFQFN])
        
      else:                          
        # else user cancelled..
        showUserCancel()
      #
    else:                            # if len(objsVisible) != 0
    # Tell user what's up
      Msg = "Nothing to export, Please select one or more solid objects;\n\n\t(NOT just 'Origin')"
      msgType = msgBox.Warning
      Btns = msgBox.Ok
      usrSel = progressDlg(macroName, Msg, Btns, msgType) 
           
  else:
  # else user cancelled..
    showUserCancel()

  

```

CIKoolK, 2025-03-24:
Ho-K, I have tested the b40928 ._AppImage_ with _Fedora Work Station 41_, _Ubuntu GNOME 46_, _Debian 12_, _MInt LMDE,_ and my _Mint v22.9/MATÉ_ installation.

It crashes with the same (Could not find the Qt platform plugin "xcb" in "") error on all...

**Did no one test this prior to release**?

JohnOCFII, 2025-03-24:
> **Did no one test this prior to release**?

Chill out.  You are talking about a **development** release.  The release process is under heavy reconstruction right now.  If you want a stable release, it is available here: https://www.freecad.org/downloads.php

CIKoolK, 2025-03-24:
> > **Did no one test this prior to release**?
> 
> Chill out. You are talking about a **development** release. The release process is under heavy reconstruction right now. If you want a stable release, it is available here: https://www.freecad.org/downloads.php
> 
> Details of the issue can be found here: [#376 (comment)](https://github.com/FreeCAD/FreeCAD-Bundle/issues/376#issuecomment-2726776482)

Oh come on!  Someone should have tested it to see if it at least launched, "development release" or not--just out of curiosity if for no other reason...

oursland, 2025-03-24:
> Ho-K, I have tested the b40928 ._AppImage_ with _Fedora Work Station 41_, _Ubuntu GNOME 46_, _Debian 12_, _MInt LMDE,_ and my _Mint v22.9/MATÉ_ installation.
> 
> It crashes with the same (Could not find the Qt platform plugin "xcb" in "") error on all...
> 
> **Did no one test this prior to release**?

Nope.  These are automated weekly releases.  If you want a tested, stable build, please use the v1.0.0 release.

CIKoolK, 2025-03-24:
> Nope. These are automated weekly releases. If you want a tested, stable build, please use the v1.0.0 release.

Then they should not be called "releases"--and made public--they are "in-house" test builds...

hyarion, 2025-03-24:
> Then they should not be called "releases"--and made public--they are "in-house" test builds...

This is copy-pasted from the website download page:
> [!NOTE]
> FreeCAD's development happens daily! The FreeCAD community generates weekly builds that are based on bleeding edge FreeCAD code in order for users to test bugfixes/regressions along with new features. We ask that advanced users occasionally run the development builds to assist with testing new code. These builds are not suitable for production use, and care should be taken when using them (back up your files regularly, etc.). Development builds should be expected to be slower, consume more memory, and be less stable than the official release versions.

CIKoolK, 2025-03-24:
They need to add: "_These are automated builds, that have not been tested in any manner, and may not even launch._"

hyarion, 2025-03-24:
I think it is already implied, but feel free to add a suggestion on the web-page repo if you think it needs to be more explicit.

hyarion, 2025-03-24:
@huguesdpdn-aerospace I think that was a bit uncalled and please avoid calling names, ok ❤️. I only pointed CIKoolK to how the webpage warns the user about the dev build and where to suggest changes 🤷‍♂ 

> [Maintainers]:
> Do we have a FreeCAD task about this? Or a tracking ticket ? I see nothing about Qt6 in https://github.com/orgs/FreeCAD/projects . Because from my point of view, this is a big thing.
> Shall we revert the last commits [@adrianinsaval](https://github.com/adrianinsaval) ?

This is a well known issue which happen because of missing communication between maintainers and packing teams.
It has been discussed if we should revert it (can't find the PR where it was discussed right now) but main has too many changes now to make it work.
Oursland is working on it and he has a PR draft on it here: https://github.com/FreeCAD/FreeCAD/pull/20415

JohnOCFII, 2025-03-24:
> Do we have a FreeCAD task about this? Or a tracking ticket ? I see nothing about Qt6 in https://github.com/orgs/FreeCAD/projects .

It was discussed and agreed to move forward during a recent Developer's meeting.  Here's a link to the minutes of that meeting: https://github.com/FreeCAD/FreeCAD-developer-meetings/blob/main/Minutes/minutes-2025-03-02.md

RWL-69, 2025-03-24:
I have the same issue.  While trying to find out details with debug I got this.
```

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
(qt.qpa.plugin) Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
```
and

```
qt.qpa.plugin: init_platform called with pluginNamesWithArguments "wayland" platformPluginPath "" platformThemeName ""
qt.core.plugin.factoryloader: checking directory path "/tmp/.mount_FreeCANlmGaA/usr/bin/platforms" ...
qt.qpa.plugin: Attempting to load Qt platform plugin "wayland" with arguments QList()
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
```

Will wait for the next weekly build to test.

adrianinsaval, 2025-03-24:
Appimage should work now

CIKoolK, 2025-03-24:
> Appimage should work now

Unfortunately my workstation's CPU cooling fan died this morning, new one comes Wednesday, so I'm down 'til sometime Wednesday evening./

adrianinsaval, 2025-03-25:
I tested and is working so I'll close this issue, you may open new issues if you find other problems in the new builds

adrianinsaval, 2025-03-25:
note that the previous qt5 builds are available at https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/devel-qt5

closed 2025-03-25 by adrianinsaval

J-Dunn, 2025-03-31:
> Nope. These are automated weekly releases. If you want a tested, stable build, please use the v1.0.0 release.

Then the automated process needs to be improved to see whether it actually starts or creates a coredump ;)

The previous comment that these should not be called "releases" if they are nothing more than an automated build script no one has even tried to run, seems valid. 

At a lower level, many parts of FreeCAD have test units. It would seem that would be a good idea here.