FreeCAD/FreeCAD-Bundle #390: Package deliveries testing are insufficient

huguesdpdn-aerospace, 2025-03-24:
On the recent events mentionned in https://github.com/FreeCAD/FreeCAD-Bundle/issues/387 , and from the comments in there, @CIKoolK raised a valid point in https://github.com/FreeCAD/FreeCAD-Bundle/issues/387#issuecomment-2746672819 : seems that automated deliveries are not enough tested

Here are some proposals: Run all binaries once in each deliveriy and:

- run `ldd` on each binary and check for no error
- `grep`/search in the output for any error message
- check that the binary is alive more than 50 seconds without crashing/closing or displaying any error message (or important warning message)
- perform a screenshot before execution and at the 30th second and check that UI is correctly display (we dont need to be precise at pixel level or get the exact same image as expected, but at least the BEFORE/AFTER binary execution are very different
- run a default script that create a very simple 3D object and check for any crash/high RAM usage/high CPU usage/broken links with libs

My guess is that the first and second checks are easy to implement
The third check needs to have a server with a display for some binaries (but would be  easy for `freecadcmd` for instance)
The fourth check could be hard and I don't know if it is even technically feasable. Would it be? I guess GitHb servers do not have a display, right ? Or is there a way to have an emulated one ?

I think that those could avoid useless deliveries and detect errors before users complain about it. It will also prove that FreeCAD is a robust software compare to some alternatives and reduce backlog.

What do you think?

CIKoolK, 2025-03-24:
#'s 1 and 2 would have caught this fundamental error--the others would be nice, however I recognize time is always limited... My last position before retirement was in Sr. IT Mgt. for a large State health department (67 counties, 141 clinics). We created all our medical billing systems in house due to the convoluted nature of public health funding. Testing was always a non-glamourous PITA, but it had to be done...

adrianinsaval, 2025-03-24:
proably what would make the most sense is to run a portion of the freecad python tests. I'll get to work on this soon