FreeCAD/FreeCAD-Bundle #391: Update AppImage python packages

leoheck, 2025-03-25:
Can we have AppImage python packages updated?
[image]

adrianinsaval, 2025-03-29:
what packages, what for, which version?

leoheck, 2025-03-29:
all of them, to the last version.

adrianinsaval, 2025-03-30:
most of them get updated regularly on conda which we use for the builds so we get them automatically so long as their compatible with the library versions we use, if you don't have any specific request with specific reasoning I'll go ahead and close this. There is no need to chase around after the latest of everything if what's current doesn't have a problem.

Also note that several of those in your screenshot are installed locally on your machine by the addon manager, not provided by the appimage so you need to update it there

closed 2025-03-30 by adrianinsaval

leoheck, 2025-03-30:
I update these every time I have to use the Package Manager. So maybe this is a problem with it.