FreeCAD/FreeCAD-Bundle #393: Latest builds

leoheck, 2025-03-29:
Is it normal to have the same version being built multiple times? This 40971 was built 3 or 4 times already.

CIKoolK, 2025-03-29:
"normal" and "FreeCAD" are often mutually exclusive....

leoheck, 2025-03-30:
Seriously, I want to understand if this is expected.

CIKoolK, 2025-03-30:
> Seriously, I want to understand if this is the expected behavior.

I can conceive of no reason why it would be desired or "expected" behaviour, it has to be a processing error (the weekly "release" process has been _wobbly_ for some time).

leoheck, 2025-03-31:
It may be due to a manually triggered process fixing pontual issues. But I think this isn't the first time at least in this last month.

hyarion, 2025-03-31:
This is not the intended behavior. I'm closing this as issues are not intended to be used for general discussions.

closed 2025-03-31 by hyarion