FreeCAD/FreeCAD-Bundle #397: Daily PPA builds broken (2025-03-26)

furgo16, 2025-04-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The daily PPA builds have been broken for all since 2025-03-26:

- https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-daily/+packages?field.name_filter=&field.status_filter=&field.series_filter=noble
- https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-daily/+packages?field.name_filter=&field.status_filter=&field.series_filter=jammy

The `focal` builds have been broken even longer, probably due to a different issue. It might be worth thinking of retiring the builds for this series:
- https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-daily/+packages?field.name_filter=&field.status_filter=&field.series_filter=focal

The cause might be https://github.com/FreeCAD/FreeCAD/pull/20372. If it is the case, the easiest thing might be to disable PCL for the Debian builds. But only a maintainer can do it.

### Full version info

```shell
1.1.0dev
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

hyarion, 2025-04-07:
> The focal builds have been broken even longer, probably due to a different issue.

We dropped support for focal (20.04) for for 1.1.0dev as focal will be unsupported when 1.1.0 gets released.
IIRC 1.0.1 will still support focal.

Regarding PLC, it was enabled specifically for Ubuntu, so it should work. @adrianinsaval do you know what might have happened here?

furgo16, 2025-04-08:
It would be great to move the Debian daily builds out of Launchpad and Debian's Salsa (GitLab instance) to GitHub:
- The same maintainer accounts would control the infrastructure. Now three separate set of accounts (freecad-maintainers on Launchpad, adrianinsaval on Salsa) are required. Not all maintainers are on all platforms I'd think that the maintainers on Launchpad do not receive any notifications.
- Contributors do not need three different accounts and learn three platforms to contribute. One of which (Launchpad) is short of obsolete.

The build could be replicated with some work. The only challenge would be to replicate the hosting of the .deb package that Launchpad provides.

furgo16, 2025-04-11:
Alternatively, should https://github.com/FreeCAD/FreeCAD/pull/20372 be reverted? From the PR description, I cannot follow the motivation for the change. Unless it provides any tangible benefits, it might be worth re-disabling the PCL build.

It also gave me trouble with the snap builds, and I had to disable it there.

hyarion, 2025-04-11:
> Alternatively, should [FreeCAD/FreeCAD#20372](https://github.com/FreeCAD/FreeCAD/pull/20372) be reverted? From the PR description, I cannot follow the motivation for the change. Unless it provides any tangible benefits, it might be worth re-disabling the PCL build.
> 
> It also gave me trouble with the snap builds, and I had to disable it there.

@adrianinsaval what's your thought about this? --^

vosk, 2025-05-01:
(copied over from the duplicate)
Hey libpcl-dev is missing from here:
https://git.launchpad.net/freecad/tree/debian/control?h=ppa/main
anyone access to change the recipe here?
https://code.launchpad.net/~freecad-maintainers/+recipe/freecad-daily

```
# git-build-recipe format 0.4 deb-version {debupstream}~{time}
*gh*/FreeCAD/FreeCAD main
nest-part debian *gh*/FreeCAD/FreeCAD-Bundle debian debian debian
```
We just need a branch with the debian build related info in this repo

furgo16, 2025-05-01:
Thanks. While the fix is known, the main issue is that the PPA is unmaintained. There is also a proposal to indirectly fix this by reverting the PR that enabled PCL in the first place: https://github.com/FreeCAD/FreeCAD/pull/20995.

vosk, 2025-05-02:
I dont understand what unmaintained really means.  I'd be happy to fix that if @adrianinsaval or @sgrogan or @abdullahtahiriyo like to add me to the launchpad maintainers, I sent a request to be added (Manos O. is my id there). I totally understand if thats not aligned with the projects policies etc.

closed 2025-05-05 by adrianinsaval