FreeCAD/FreeCAD-Bundle #403: AppImage 1.1.0 can't be integrated and started in LMDE 6 Cinnamon

Dan-cer, 2025-05-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have successfully installed the latest AppImage in Linux Mint 22.1 using the AppImageLauncher. 
However, in Linux Mint Debian Edition - LMDE 6 - there is only an error message:

> Error when registering the AppImage in the system via libappimage

I attached a screenshot. 
Starting the AppImage via Terminal I got this output:

`rainer@Optiplex:~/Applications$ ./FreeCAD_1.0.0-conda-Linux-x86_64-py311_762d03e714090fee85c2cd69af722050.AppImage
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_shall_not_be_integrated : sqfs_open_image error: /home/rainer/Applications/FreeCAD_1.0.0-conda-Linux-x86_64-py311_762d03e714090fee85c2cd69af722050.AppImage
AppImageLauncher error: appimage_shall_not_be_integrated() failed (returned -1)
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_is_terminal_app : sqfs_open_image error: /home/rainer/Applications/FreeCAD_1.0.0-conda-Linux-x86_64-py311_762d03e714090fee85c2cd69af722050.AppImage
AppImageLauncher error: appimage_is_terminal_app() failed (returned -1)

(AppImageLauncher:16005): Gtk-WARNING **: 14:53:04.282: Error loading theme icon 'dialog-question' for stock: Schwerer Fehler beim Lesen einer PNG-Bilddatei: bad parameters to zlib

(AppImageLauncher:16005): Gtk-WARNING **: 14:53:04.349: Error loading theme icon 'dialog-question' for stock: Schwerer Fehler beim Lesen einer PNG-Bilddatei: bad parameters to zlib
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_register_in_system : sqfs_open_image error: /home/rainer/Applications/FreeCAD_1.0.0-conda-Linux-x86_64-py311_762d03e714090fee85c2cd69af722050.AppImage

(AppImageLauncher:16005): Gtk-WARNING **: 14:53:07.864: Error loading theme icon 'dialog-error' for stock: Schwerer Fehler beim Lesen einer PNG-Bilddatei: bad parameters to zlib
Error in file "/home/rainer/.local/share/applications/Pdfxedit.cinnamon-generated.desktop": "" is an invalid MIME type ("" does not contain a subtype)
Error in type 'unknown/unknown' (in /home/rainer/.local/share/mime/packages/install4j_21zhne-application.pdf;application.x-gzpdf;application.x-bzpdf;application.x-xzpdf;application.x-ext-pdf.xml): Invalid MIME-type 'application/pdf;application/x-gzpdf;application/x-bzpdf;application/x-xzpdf;application/x-ext-pdf'.`

I tried it without success both with the stable version and daily builds.

[image]

### Full version info

```shell
Not possible because FreeCAD could't start.
However, I used the AppImage in LM 22.1 Cinnamon too. The version information is:

OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)  | OS: LMDE 6 Cinnamon
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

CIKoolK, 2025-05-16:
AppImageLauncher. has not been updated in 4 years, it does not use the latest compression algorithm used by the AppImage packaging application.

Just give the FC .AppImage package execution permission and launch it. [It runs on LMDE quite well](url=https://www.paladinmicro.com/RedditStuff/FreeCAD-LMDE-00.png)...

Dan-cer, 2025-05-16:
I tried that without success. Unfortunately I can't investigate this issue any longer bc I switched OS from LMDE 6 to LM 22.1 (both Cinnamon) for other reasons.

closed 2025-05-16 by Dan-cer