FreeCAD/FreeCAD-Bundle #405: AppImage: Links on .md pages are not clickable, missing gio-launch-desktop

nazar-pc, 2025-05-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Links on pages rendered in Addons Manager (likely from Markdown) are not clickable

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37709 (Git) AppImage
Build type: Release
Branch: main
Hash: dd816a635ccacc39f29b1761c6ccfed21a90b646
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

oursland, 2025-05-10:
The `conda-forge` does not provide the files `gio-launch-desktop` nor `gio` in recent releases of `glib`.

The package metadata tool may be used to explore which packages provide which files: https://conda-metadata-app.streamlit.app/Search_by_file_path

adrianinsaval, 2025-05-12:
is this still a problem in the qt6 builds? I thought this was a qt5 bug

nazar-pc, 2025-05-13:
Works for me in latest weekly:
```
OS: Ubuntu 24.04.2 LTS (Custom/custom/xcb)
Architecture: x86_64
Version: 1.1.0dev.41628 (Git) AppImage
Build date: 2025/05/09 10:50:32
Build type: Release
Branch: main
Hash: 9577d108e333c80ee640211271a76105f52baf14
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Ukrainian/Ukraine (uk_UA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/83.3641/1.25
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * Curves 0.6.42
```

closed 2025-05-13 by maxwxyz