FreeCAD/FreeCAD-Bundle #406: [Problem]Weekly built on MacOs High sierra is unable to run

Marco5878, 2025-05-08:
Bug description. 
MacOs 10.13.6  Issue the following message when launching Freecad last weekly built using the bin executable inside package.
  
dyld: Symbol not found: _NSAppearanceNameDarkAqua
  Referenced from: /Applications/Utilities/FreeCADr.app/Contents/Resources/bin/../lib/libQt6Core.6.dylib (which was built for Mac OS X 11.0)
  Expected in: /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit
 in /Applications/Utilities/FreeCADr.app/Contents/Resources/bin/../lib/libQt6Core.6.dylib
Abort trap: 6

Seems to make reference to a lib that does not exist under MacOs 10.13.6
**To Reproduce**
Steps to reproduce the behavior:
1. Standard launch

**Screenshots**
Popup message on screen at standard launch

[image]

**Additional context**
IMac 27" Mid-2011 Mac Os High sierra 10.13.6

oursland, 2025-05-10:
Looks like the Qt6 dependency is no longer built for macOS versions under 11.0 since https://github.com/conda-forge/qt-main-feedstock/commit/88bdec651e3b412f3821c9e354fc02c400cce96b

@adrianinsaval @chennes

adrianinsaval, 2025-05-11:
what should be our guideline for macos support? 10.13 is 7y old and 4y past it's EOL according to https://endoflife.date/macos

hyarion, 2025-05-11:
> what should be our guideline for macos support? 10.13 is 7y old and 4y past it's EOL according to https://endoflife.date/macos

As I understand this; in this case it's not that we drop support for a specific version of our own choice, it's that conda-forge does it - which our build depends upon.
Users of old versions can still build it if they get their deps from other sources.

JohnOCFII, 2025-05-11:
> what should be our guideline for macos support? 10.13 is 7y old and 4y past it's EOL according to https://endoflife.date/macos

Ideally, we'd use data from telemetry to see where most of the users are -- but Telemetry is new and opt-in, so that likely won't give a complete picture.  In other projects I've seen teams settle on something that looked like this: "Current + former major releases, plus not actively removing support for releases older than that, but just accepting that they won't be supported when key ingredients (such as the build tools, major libraries like QT, etc.) are no longer supported in the combinations required to build FreeCAD.

So today, that'd include macOS 15 Sequoia(current major release), macOS 14 Sonoma (previous major release) and not automatically removing support for anything older, but just dropping it when components to build FreeCAD begin to fail.

For reference, macOS 10.14 Mohave, first released in June 2018, and last updated in July 2021 was the last release to support 32-bit Intel chips.  All since then have required 64-bit either Intel or Arm.

This table from WikiPedia is pretty good, showing names, dates, etc. https://en.wikipedia.org/wiki/MacOS_version_history

oursland, 2025-05-12:
@ipatch is the Homebrew-based build functional for OS 10.x?

Marco5878, 2025-05-12:
This Mac is still very good at doing the job for most application ( for intel based) but it  cannot be upgraded beyond High Sierra 10.13.6 It is a 64 bis core. 
if the build for 10.13 cannot be done i may consider using a patch like dosdude. 

[image]

oursland, 2025-05-12:
Unfortunately, it appears that dosdude1's patcher does not support Big Sur (macOS 11), which is the minimum version that the dependency software has been built for.  It may be possible to build FreeCAD locally on your machine using the Homebrew dependencies, but they are constantly shifting resulting in an unstable environment.  @ipatch is the maintainer of the Homebrew managed dependency builds, so hopefully he can chime in with guidance on whether or not that machine is supported.

adrianinsaval, 2025-05-12:
https://docs.brew.sh/Installation#2
> macOS 13 (Ventura) or higher is best and supported, 10.11 (El Capitan) – 12 (Monterey) are unsupported but may work and 10.10 (Yosemite) and older will not run Homebrew at all.

Maybe the best course of action for MacOS 10.x users is to pin qt6 to 6.6.0 in pixi.toml, update the lockfile and build themselves

oursland, 2025-05-12:
I see that Qt6 6.6.0 was the last version built by the feedstock compatible with macOS 10.15, which is still higher than what is being requested.  According to [the official documentation](https://doc.qt.io/archives/qt-6.6/macos.html) Qt 6.6.0 was also intended to target macOS 11.

Looking through the `qt-main-feedstock`'s `qt6` branch, it appears that no Qt 6 releases were compatible with macOS 10.13.  Only Qt5 was compatible with that platform.

adrianinsaval, 2025-05-13:
Right, though luck then. I think we're way past the expected support window for 10.x though so I'm going to close this as won't fix. Try your luck with homebrew

closed 2025-05-13 by adrianinsaval