FreeCAD/FreeCAD-Bundle #428: [Problem]Some functions very slow: query regarding level of debugging code in pre-release versions

RalphBacon, 2025-08-04:
This is not a bug report per se, but a query regarding whether I should be reporting slow functionality as a bug.

Can someone confirm the level of debugging code left in pre-release versions of FreeCase? 

If there is a large amount, or it was built in debugging (not release) mode then that might explain why, for example, using any kind of fillet in my (simple) designs causes a 30 second to 2 minute delay as things are "recomputed".

At this stage just a query for devs to confirm.

gianlucafiore, 2025-08-05:
Look, buddy, it happens to me, for example, that when drawing draft lines there is no longer any slowness, but when drawing BIM walls they do become slow.

Just as I reported in another ticket, when drawing the first wall, the graphic moves at full speed. And when confirming the coordinates, it pauses or breaks, and the following walls are drawn with a lag in the graphic interaction.

Steps to reproduce:
- Start Freecad from a closed state.
- Draw any draft object, preferably a wall, in the completely empty file.
- A 2-second wait occurs after the program freezes.
- The walls will then work, but more slowly, with a kind of lag when moving the cursor while drawing. @Syres916

RalphBacon, 2025-08-08:
That's not quite what I am experiencing.

My delay occurs whilst the model recomputes, the more complex the model, or one with fillets, I can wait for 30 seconds to 2 minutes whilst it does its stuff. On a 12-core, 4GHz computer! Crazy.

I have to go to the model and select "skip recomputes" just to be able to use FreeCAD sometimes. Or "suppress" the fillet tree tentry (or troublesome feature) until I am pretty much done.

I will download the new 1.0.2 and see if it exists there too; if so I will raise a bug.